f32c/arduino #1: Problem with compiling sketch in linux

goran-mahovlic, 2015-07-08:
Just did the example SerialEvent with baudrate changed to 19200. Here is the output:

/home/magnus/.arduino15/packages/FPGArduino/tools/f32c-compiler/1.0.0/bin/mips-elf-ar rcs /tmp/build1860868207851946728.tmp/core.a /tmp/build1860868207851946728.tmp/UARTClass.cpp.o 
/home/magnus/.arduino15/packages/FPGArduino/tools/f32c-compiler/1.0.0/bin/mips-elf-ld -N -EL -gc-sections --library-path=/home/magnus/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/system/mips/lib -nostdlib --section-start=.init=0x400 -lcrt0 /tmp/build1860868207851946728.tmp/mySerialEvent.cpp.o /tmp/build1860868207851946728.tmp/core.a -lulx2s -lm -lc -o /tmp/build1860868207851946728.tmp/mySerialEvent.cpp.elf 
/tmp/build1860868207851946728.tmp/mySerialEvent.cpp.o: In function `_GLOBAL__sub_I_inputString':
mySerialEvent.cpp:(.text.startup._GLOBAL__sub_I_inputString+0x28): undefined reference to `__dso_handle'
mySerialEvent.cpp:(.text.startup._GLOBAL__sub_I_inputString+0x2c): undefined reference to `__cxa_atexit'
/tmp/build1860868207851946728.tmp/core.a(WString.cpp.o): In function `String::changeBuffer(unsigned int)':
(.text._ZN6String12changeBufferEj+0x1c): undefined reference to `realloc'
/home/magnus/.arduino15/packages/FPGArduino/tools/f32c-compiler/1.0.0/bin/mips-elf-ld: /tmp/build1860868207851946728.tmp/mySerialEvent.cpp.elf: hidden symbol `__dso_handle' isn't defined
/home/magnus/.arduino15/packages/FPGArduino/tools/f32c-compiler/1.0.0/bin/mips-elf-ld: final link failed: Bad value
Error compiling.

goran-mahovlic, 2015-07-08:
Trying this workaround 
put this line somewhere in your sketch.
void* __dso_handle;
from 
http://www.esp8266.com/viewtopic.php?p=13056

gives another error 

/home/magnus/.arduino15/packages/FPGArduino/tools/f32c-compiler/1.0.0/bin/mips-elf-ld -N -EL -gc-sections --library-path=/home/magnus/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/system/mips/lib -nostdlib --section-start=.init=0x400 -lcrt0 /tmp/build1860868207851946728.tmp/mySerialEvent.cpp.o /tmp/build1860868207851946728.tmp/core.a -lulx2s -lm -lc -o /tmp/build1860868207851946728.tmp/mySerialEvent.cpp.elf 
/tmp/build1860868207851946728.tmp/mySerialEvent.cpp.o: In function `_GLOBAL__sub_I_inputString':
mySerialEvent.cpp:(.text.startup._GLOBAL__sub_I_inputString+0x2c): undefined reference to `__cxa_atexit'
/tmp/build1860868207851946728.tmp/core.a(WString.cpp.o): In function `String::changeBuffer(unsigned int)':
(.text._ZN6String12changeBufferEj+0x1c): undefined reference to `realloc'
Error compiling.

gornjas, 2015-07-08:
I've just commited a crude realloc() implementation and rebuilt the C libs. With the new libc I can get the SerialEvent example to compile by adding the following code chunk to the sketch:

void *__dso_handle;
__BEGIN_DECLS
int __cxa_atexit(void (*destructor) (void *), void *arg, void *dso) { return (0);}
__END_DECLS;

I doubt this will work since we haven't actually implemented serialEventRun() anywhere. In standard Arduino libs this seems to be implemented in HardwareSerial.cpp. We could probably implement this in UARTClass.cpp but looking at the standard Arduino code I'm having a hard time guessing the exact semantics of serialEvent stuff.

goran-mahovlic, 2015-07-09:
Got Hx711 Scale module working
I used this workaround to fix same error, files I have changed are here

https://github.com/goran-mahovlic/f32c_extras/blob/master/Arduino/MimasV2/SerialScale/SerialScale.ino
https://github.com/goran-mahovlic/f32c_extras/blob/master/Arduino/MimasV2/libs/Hx711/Hx711.cpp

emard, 2015-07-12:
I placed the __dso_handle fix to SoftwareWire for now.

We may solve the problem cleanly

If it arires with another library then we can
put this fix somewhere in the core

goran-mahovlic, 2015-07-15:
I will close this issue for now, and reopen if preoblem arries...

emard, 2017-11-06:
The problem reappears when compiling MIDI library
With including this into the sketch, I got it to compile

#include <MIDI.h>
// drty workaround to get MIDI compiling
void* __dso_handle = (void*) &__dso_handle;
__BEGIN_DECLS
int __cxa_atexit(void (_destructor) (void *), void *arg, void *dso) { return (0);}
__END_DECLS;

emard, 2017-11-22:
Above is fixed in abi.gcc but there's new one

int test(int a)
{
  static int b = a + 1;
  return b;
}

will abort with error
/tmp/arduino_build_186372/sketch/Blink.ino.cpp.o: In function `test(int)':
(.text._Z4testi+0x20): undefined reference to `__cxa_guard_acquire'
(.text._Z4testi+0x38): undefined reference to `__cxa_guard_release'

emard, 2017-11-22:
this error may be fixed in file platform.txt by adding compiler
option -fno-threadsafe-statics
into compiler.cpp.flag lines:
f32c.mips.compiler.cpp.flags
f32c.riscv.compiler.cpp.flags