f32c/arduino #2: Support for Software I2C

goran-mahovlic, 2015-07-08:
I have done some research, and find library that should work like wire library
https://github.com/Testato/SoftwareWire
tested oled on Arduino UNO on pins 2 and 3, and it is working
FPGAarduino returns errors on compiling Small_example.ino

goran-mahovlic, 2015-07-09:
error: 'F_CPU' was not declared in this scope
solved by adding to SoftWire.cpp
#define F_CPU 100000000UL // 100 MHz

Ser.println( (uint16_t)
replaced with
Ser.println( (size_t)

now only few errors remain 

C:\Users\gmahovlic\Downloads\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\libraries\SoftwareWire\SoftwareWire.cpp: In constructor 'SoftwareWire::SoftwareWire(uint8_t, uint8_t, boolean, boolean)':
C:\Users\gmahovlic\Downloads\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\libraries\SoftwareWire\SoftwareWire.cpp:141:8: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   port = digitalPinToPort(_sdaPin);
        ^
C:\Users\gmahovlic\Downloads\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\libraries\SoftwareWire\SoftwareWire.cpp:144:39: error: 'portModeRegister' was not declared in this scope
   _sdaDirReg   = portModeRegister(port);
                                       ^
C:\Users\gmahovlic\Downloads\arduino-1.6.5-r2-windows\arduino-1.6.5-r2\libraries\SoftwareWire\SoftwareWire.cpp:147:8: error: cannot convert 'volatile uint32_t* {aka volatile unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment
   port = digitalPinToPort(_sclPin);

goran-mahovlic, 2015-07-09:
one by one changed uint8_t to uint32_t

only one error left

SoftwareWire.cpp: In constructor 'SoftwareWire::SoftwareWire(uint8_t, uint8_t, boolean, boolean)':
SoftwareWire.cpp:144:39: error: 'portModeRegister' was not declared in this scope
   _sdaDirReg   = portModeRegister(port);
                                       ^
Error compiling

emard, 2015-07-09:
Let's fix a bit arduino core wiring_digital.h file 
so it can compile this stuff more cleanly

It can solve most of stuff except this uint8_t * should be manually
converted to something like (volatile uint32_t *)

Seems mission the portModeRegister function as it seems mission
and to boards.txt this might solve F_CPU 
build.f_cpu=<freq_in_Hz>

If neccessary for support, portmode, gpio.vhd module can also be modified
to support this semantics

gornjas, 2015-07-09:
F_CPU should never be hardcoded, but rather defined to equal to VARIANT_MCK, since VARIANT_MCK is already properly defined for a particular CPU configuration by pulling the right value from {build.fcpu} variable propagated from the Arduino IDE.

emard, 2015-07-09:
F_CPU is there in the git and seems working

goran-mahovlic, 2015-07-12:
Confirmed I2C is working now...

closed 2015-07-12 by goran-mahovlic