f32c/arduino #4: undefined reference to `shiftOut'

goran-mahovlic, 2015-07-12:
if I try to use simple code for nokia display I got this error 
I have seen this error on multiply examples

http://playground.arduino.cc/Code/PCD8544

goran-mahovlic, 2015-07-12:
adding this to ino works 
but I do now know where to place it

void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val)
{
     uint8_t i;

     for (i = 0; i < 8; i++)  {
           if (bitOrder == LSBFIRST)
                 digitalWrite(dataPin, !!(val & (1 << i)));
           else      
                 digitalWrite(dataPin, !!(val & (1 << (7 - i))));
                 
           digitalWrite(clockPin, HIGH);
           digitalWrite(clockPin, LOW);            
     }
}

gornjas, 2015-07-12:
There's already an implementation in hardware/arduino/avr/cores/arduino/wiring_shift.c. Perhaps this could work, most probably even without any modifications if copied to hardware/fpga/f32c/cores/f32c?

goran-mahovlic, 2015-07-12:
now I got I probably need to fix .h file

/home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.c:27:9: error: conflicting types for 'shiftIn'
 uint8_t shiftIn(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder) {
         ^
In file included from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/Arduino.h:80:0,
                 from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_private.h:30,
                 from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.c:25:
/home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.h:26:17: note: previous declaration of 'shiftIn' was here
 extern uint32_t shiftIn( uint32_t ulDataPin, uint32_t ulClockPin, uint32_t ulBitOrder ) ;
                 ^
/home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.c:42:6: error: conflicting types for 'shiftOut'
 void shiftOut(uint8_t dataPin, uint8_t clockPin, uint8_t bitOrder, uint8_t val)
      ^
In file included from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/Arduino.h:80:0,
                 from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_private.h:30,
                 from /home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.c:25:
/home/mistery/.arduino15/packages/FPGArduino/hardware/f32c/1.0.0/cores/f32c/wiring_shift.h:32:13: note: previous declaration of 'shiftOut' was here
 extern void shiftOut( uint32_t ulDataPin, uint32_t ulClockPin, uint32_t ulBitOrder, uint32_t ulVal ) ;
             ^
Error compiling.

goran-mahovlic, 2015-07-12:
copy wiring_shift.c to hardware/fpga/f32c/cores/f32c, and changed all  uint8_t to uint32_t in wiring_shift.h

and got it working without errors, so you can fix it on repo

goran-mahovlic, 2015-07-12:
and now 
https://github.com/wayoda/LedControl
is working to :)

goran-mahovlic, 2015-07-12:
all is working now with json

closed 2015-07-12 by goran-mahovlic