f32c/arduino #6: No serial output

goran-mahovlic, 2015-07-16:
This simple example does not return serial

void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(19200);
}

void loop() {
  Serial.print("1");
  Serial.print("2");
  Serial.print("3");
  Serial.print("4");
  Serial.println("5");
}

and this one is working

  // initialize serial communication at 9600 bits per second:
  Serial.begin(19200);
}

void loop() {
  Serial.print("1");
  Serial.print("2");
  Serial.print("3");
  Serial.println("4");
}

emard, 2015-07-16:
Both examples work on ulx2s
You probably use HEX protocol (blind upload, no CRC) 
and noisy serial port which.

It is essential to get BIN protocol in ujprog working on your board
because it has CRC and assures clean upload.

ujprog is tested only on 115200, and may require tweaking the 
timeouts for 19200

goran-mahovlic, 2015-08-13:
and now with 115200 and with binary CRC upload serial is still not working good

when I try example 

void setup() {
// initialize serial communication at 9600 bits per second:
Serial.begin(115200);
}

void loop() {
Serial.print("1");
Serial.print("2");
Serial.print("3");
Serial.print("4");
Serial.println("5");
}

I got only one response from serial and it breaks and stops sending

emard, 2015-08-13:
> void loop() {
> Serial.print("1"); delay(1000);
> Serial.print("2"); delay(1000);
> Serial.print("3"); delay(1000);
> Serial.print("4"); delay(1000);
> Serial.println("5"); delay(1000);
> }
>
> I got only one response from serial and it breaks and stops sending
>
> ---

> and now with 115200 and with binary CRC upload serial is still not working good
> 
> when I try example 
> 
> void setup() {
> // initialize serial communication at 9600 bits per second:
> Serial.begin(115200);
> }

if this with delay is working then
f32c is probably OK, but usbserial firmware
has problems keeping up with "high" speed of
115200 baud

goran-mahovlic, 2015-12-03:
I have tested, and serial is working normaly under linux, but on windows it is not working, I have talked with emard and he sugested to refresh USB to serial after loading sketch. I have found microsoft tool that do this from console it is called devcon
after disabling and enabling USB serial is working...
It is hard to get this tool

http://www.robvanderwoude.com/devcon.php
http://techlikes.com/2010/09/25/devcon-problem-in-windows-7-solved.html
http://karuppuswamy.com/wordpress/2010/06/05/how-to-plug-in-and-plug-out-usb-device-through-command-line-in-windows/

after I did this serial started to work
https://yadi.sk/i/2S0jrrtVkvvak

Is there any way that we automatize two commands ...

emard, 2015-12-03:
Stavio sam to u github arduino ali nije
jos u automatskom.

Meni ga poziva sa ispravnim parametrima
barem na 1vi pogled - treba probat za stvarno

glavni fajl koji definira pathove i commandline
je boards-manager/boards.txt

skopiraj ga ručno te iz arduina
pokreni burn bootloader, vidjeces
errore i commandliniju koju pozove
pa mu tamo gdje treba dodaj
python i bitstream

closed 2015-12-04 by emard