f32c/arduino #9: MIPS gcc won't compile u8g, RISC-V compiles it

emard, 2015-09-14:


emard, 2016-01-01:
Here is a simple file that manifests this error
I have seen it on gcc 4.8 on linux 64-bit and it looks
like linker has some bug of wrong combination 
of 32 and 64 bit integer arithmetic.

Workaround: 
in f32c bitstream remap RAM to  address below 0x80000000 e.g. 0x10000000 
in boards.txt for your board change something like
fpga_esa11_7a35i_bram.menu.ramsize.32M.compiler.ld.extra_flags=--section-start=.init=0x80000000
to
fpga_esa11_7a35i_bram.menu.ramsize.32M.compiler.ld.extra_flags=--section-start=.init=0x10000000

[error_manifest.txt](https://github.com/f32c/arduino/files/75888/error_manifest.txt)

updating the compiler would be nice too  :)

gornjas, 2016-01-14:
This should mask the ld bug: https://github.com/f32c/f32c/commit/52d0fc0eb1c5afed666a24502b977d6fb7afdd21

emard, 2016-01-15:
found mips linker bug, this workaround fixes it
objcopy -R .rel.dyn
u8g and error_manifest now compile properly
haven't tested the u8g display myself but it
should work now as well as for RISC-V

closed 2016-01-15 by emard