f32c/arduino #19: Incorrect library

Acridka, 2016-02-18:
Hello!
I have found ur library about 2 days and I have some problems with it.
I am using Spartan Start Kit S3E-500. I done all you said to see my board, I see it on Board Manager, and I see ujprog, so it seems to be correct.But when I try to compile any example I have a error (it is on white colour so it seems to be warning may be) `incorrect library found in C:\Users\Acidka\AppData\Local\Arduino15\packages\FPGArduino\hardware\f32c\1.0.0\libraries\f32c: C:\Users\Acidka\AppData\Local\Arduino15\packages\FPGArduino\hardware\f32c\1.0.0\libraries\f32c`
And there was some problems with paths on this path like bootloader foulder was in "tools" and he couldn't find it, but I replaced and he found. So if I try to load it it just do nothing.

Thanks for help,
Acridka.

Acridka, 2016-02-18:
it repairs by use 1.6.4 or 1.6.5 arduino ide for this . . .

closed 2016-02-18 by Acridka

emard, 2016-02-18:
OK, the bootloader (actually it contains fpga bitstream with whole f32c
system)
is quite old for the spartan-3 starter kit version,
I have long time not recompiled. However it used to work and still
should be fine to blink and fade leds

Of course it doesn't have new features like for example textmode video
or FM rds encoder but it should work with a bit of project
fixing (adding new vhdl files from more recently maintained projects
see ulx2s or scarab) and recompile