f32c/arduino #32: Add picorv32 support for Alhambra board

drtrigon, 2018-10-22:
According to https://github.com/cliffordwolf/picorv32/issues/92 picorv32 RISC soft core runs on Alhambra board. Is it possible to add support for this to FPGArduino?

emard, 2018-10-22:
If picorv32 core can use the same gcc that works for our f32c-rv32
and approximately have the SOC (mmaped io, leds, spi, timers)
also same or similar to f32c soc and are mapped preferrably
to the same memory addresses then I'd be happy to add
some board names, strings and compiler options to boards.txt
and/or platform.txt

Can somebody try our rv32 binaries on picorv32 and
report if they work at least to make a led blinky?

drtrigon, 2018-10-23:
I try to help with whatever I can. The toolchain I used successfully is "RISC-V GNU toolchain and libraries for a pure RV32I target" as mentioned in https://github.com/cliffordwolf/picorv32#building-a-pure-rv32i-toolchain. It uses `riscv32-unknown-elf-gcc`, `riscv32-unknown-elf-objcopy` and from icestorm/icestudio toolchain `iceprog`.

emard, 2018-10-23:
Can you try to use our f32c-rv32 compiled binary to get LED blinky on picorv32?
In the meantime I probably should get picorv32 for ulx3s, I think it's already
done at prjtrellis but haven't tried to download and compile

I'm currently busy with new boards ULX3S and some hardware support around
it

drtrigon, 2018-10-26:
> Can you try to use our f32c-rv32 compiled binary to get LED blinky on picorv32?

I tried, but not with much success... I used code from a working project and just tried to replace the toolchain with yours, see attached [Makefile.txt](https://github.com/f32c/arduino/files/2520503/Makefile.txt) (the commented parts are the ones from the working toolchain for reference).

Running it results in (the .elf file has a size of 0 bytes):
> $ make prog
../../fpgarduino/f32c/bin/riscv-elf-gcc -std=gnu11 -DHX8KDEMO -march=RV32I -Wl,-Bstatic,-T,sections.lds,--strip-debug -ffreestanding -nostdlib -o firmware.elf start.s firmware.c
collect2: error: ld terminated with signal 11 [Segmentation fault], core dumped
$HOME/Downloads/icestudio/fpgarduino/f32c/bin/../lib/gcc/riscv-elf/4.9.2/../../../../riscv-elf/bin/ld: /tmp/cc4XMh7D.o: ABI is incompatible with that of the selected emulation
$HOME/Downloads/icestudio/fpgarduino/f32c/bin/../lib/gcc/riscv-elf/4.9.2/../../../../riscv-elf/bin/ld: failed to merge target specific data of file /tmp/cc4XMh7D.o
$HOME/Downloads/icestudio/fpgarduino/f32c/bin/../lib/gcc/riscv-elf/4.9.2/../../../../riscv-elf/bin/ld: /tmp/cc39QdcM.o: ABI is incompatible with that of the selected emulation
$HOME/Downloads/icestudio/fpgarduino/f32c/bin/../lib/gcc/riscv-elf/4.9.2/../../../../riscv-elf/bin/ld: failed to merge target specific data of file /tmp/cc39QdcM.o
make: *** [firmware.elf] Fehler 1

So my questions:
- Can you be more specific what I should test exactly (source files, commands for compiling, etc.)?
- Would it be possible to integrate the picorv32 icestorm/icestudio toolchain as it is open source too (e.g. the programmer could be needed)?

emard, 2018-10-26:
Aiieeee segmentation fault in riscv linker :)
Our riscv support is obviously not too functional
no more testing from you needed, we must
refresh this riscv support.

Maybe better for alhambra board is to use another
more stable riscv toolchain that you know works for your hardware

It is no problem for integrate from licensing point of view but
we lack of ing-hours to assemble all parts into stuff that works.
I can package another riscv compiler and all required programming
tools and libs, if you succeed with anything that leads to a blink led

gornjas, 2018-10-26:
Our current riscv toolchain is exactly the same as the mips version, i.e. is gcc-7.2 based and should be perfectly fine. The question is why the OP has fetched the long obsoleted 4.9.2 toolchain - could that be a bug in our Arduino json / packaging?

gornjas, 2018-10-27:
How did that regression happen? I thought we already switched over to 7.2 a year ago, and were using gcc-6 based toolchains prior to that. The bm/package_fpga_index.json is clearly referencing 7.2 based toolchains.

I'm inclined to wipe out all the pre-7.2 .zip archives from the web, otherwise we'll be getting into this again and again...

emard, 2018-10-27:
There were no regression, we only upgraded mips to 7.2
and riscv 7.2 I forgot :)

gornjas, 2018-10-27:
How's that possible? There's a single integrated toolchain archive for both mips + riscv toolchains per OS/architecture (win32, osx, lin32, lin64)? I.e. they are not downloaded separately, and there's no way to separate them in the json?

gornjas, 2018-10-27:
I removed the pre gcc-7.2 toolchains from the bm on our website as a precaution for this to reocur.

emard, 2018-10-27:
> When I look better, we should have 7.2 toolchain for both mips+riscv
> in boards manager for bsd linux win osx
> so gcc 4.9 probably remained from some old installation?

OK I found it
linux 32-bit is really old toolchain from 2015 because Marko provided
only linux-64 bit 7.2
and I had no better idea than to keep old toolchain for 32-bit systems
otherwise they
would compiletely break without any toolchain

drtrigon, 2018-10-27:
I got the toolchain from http://www.nxlab.fer.hr/fpgarduino/ and it was 64-bit:

- http://www.nxlab.fer.hr/dl/lin/64/ujprog
- http://www.nxlab.fer.hr/fpgarduino/bm/f32c-lin64-toolchain.tar.gz

gornjas, 2018-10-27:
The 7.2-based toolchain for 32-bit linux was built at the same time the 64-bit version was. We have 5 different toolchains based on gcc-7.2 since 2017 so there's no reason to use anything else:

login% ls -l *tool*
-r--r--r--  1 marko  fpgasvn  68284749 Nov 15  2017 f32c-fbsd64-toolchain-7_2.tar.gz
-r--r--r--  1 marko  fpgasvn  79048039 Nov 15  2017 f32c-lin32-toolchain-7_2.tar.gz
-r--r--r--  1 marko  fpgasvn  72986821 Nov 14  2017 f32c-lin64-toolchain-7_2.tar.gz
-rw-r--r--  1 marko  fpgasvn  82335149 Nov 24  2017 f32c-osx-toolchain-7_2.tgz
-r--r--r--  1 marko  fpgasvn  93959677 Nov  9  2017 f32c-win32-toolchain-7_2.zip

drtrigon, 2018-10-27:
> It is no problem for integrate from licensing point of view but we lack of ing-hours to assemble all parts into stuff that works. I can package another riscv compiler and all required programming tools and libs, if you succeed with anything that leads to a blink led

As explained in https://github.com/cliffordwolf/picorv32/issues/92 you can find a working example in https://github.com/FPGAwars/Alhambra-II-FPGA/tree/master/examples/picorv32/picosoc (see e.g. https://github.com/FPGAwars/Alhambra-II-FPGA/blob/master/examples/picorv32/picosoc/Makefile). 
The toolchain used can be found in https://github.com/cliffordwolf/picorv32#building-a-pure-rv32i-toolchain (risc-v gcc) and https://github.com/FPGAwars/toolchain-icestorm/releases/ (iceprog).
Do you need more info? More specific details?

gornjas, 2018-10-27:
> > $ make prog
> > ../../fpgarduino/f32c/bin/riscv32-elf-gcc -DHX8KDEMO -march=rv32imc -Wl,-Bstatic,-T,sections.lds,--strip-debug -ffreestanding -nostdlib -o firmware.elf start.s firmware.c
> > cc1: error: requested ABI requires -march to subsume the 'D' extension
> > make: *** [firmware.elf] Fehler 1

Hi drtrigon,

pls. try adding -mabi=ilp32 to riscv32-elf-gcc invocation options.

drtrigon, 2018-10-27:
Works! The created .elf file differs by one byte from the one the original toolchain generated a few days before. Nice! Uploaded using `iceprog`.

drtrigon, 2018-10-28:
Do you need more info from my side?

emard, 2018-10-28:
HI

I think that's ok for now, I'm preparing to update arduino release with
riscv 7.2 compiler for linux-32bit, I'm preparing some additional
files there too

drtrigon, 2018-10-28:
Oh, by the way the links on http://www.nxlab.fer.hr/fpgarduino/ are still broken, quite some confusion with "_", "-" and "." going on there, e.g.:
- wrong: http://www.nxlab.fer.hr/fpgarduino/bm/f32c-lin64-toolchain_7.2.tar.gz
- correct: http://www.nxlab.fer.hr/fpgarduino/bm/f32c-lin64-toolchain-7_2.tar.gz

emard, 2018-11-09:
HI

I'm still fixing the flash/sd bootloader to make the f32c load
at power-on the user-compiled sketch (*.bin from arduino).

I have some positive results on ulx3s and papilio-pro but in
the meantime ulx2s stopped working so I'm fixing it now.

drtrigon, 2018-11-09:
Sorry to hear that but thanks a lot for your work! (I'm eager to test it on Alhambra... ;)

emard, 2018-12-04:
> What is the current status? Do you have an estimate by when I can start
> using it on the Alhambra?

HI

q3k managed to compile f32c on ice40, maybe you can try the same:
http://www.twipu.com/RadionaOrg/tweet/1068227927773536256

Although fpgarduino can run on bare-bones risc-v for blink,
using f32c is good for it has rich SOC which would provide you a lot of
additional soft-core hardware.

drtrigon, 2018-12-23:
Tryed to contact @q3k but was not able due to a flaw in my mail prog. May be he can comment here on the details of how he "compile[d] f32c on ice40"?

emard, 2018-12-23:
I know, sometimes q3k just doesn't read mails but the
fact that f32c is possible on ICE40 is worth enough to
try

The big thing about FPGA Arduino is the SOC coming with
f32c. Many projects around offer free risc-v core and
a minimalistic SOC with a serial port and pins to blink LED

No HDMI video, FPU, DMA audio, advanced timers,
SDR radio, SPI etc...

q3k, 2018-12-23:
I got f32c to synthesize on iCE40 by hacking on the VHDL source until it synthesized via Yosys with built-in Verific support (a commercial VHDL elaboration frontend), then P&Rd it using NextPNR. I haven't tested it on HW though, as I didn't have an iCE40 devboard handy at the time. I can dig up my VHDL changes if you're curious.

And, contrary to popular belief, I _do_ read all of my email, just don't have the willpower to act upon all of them :/.

emard, 2018-12-23:
Great work q3k, it would be great if you could clone the f32c
and on the cloned f32c commit your changes that allow it to compile.

drtrigon, 2018-12-23:
I can do tests on my hardware as I own a https://github.com/FPGAwars/Alhambra-II-FPGA

drtrigon, 2018-12-25:
@q3k: If you don't commit your changes it would be great if you "can dig up [my/your] VHDL changes if you're curious" and post them. I'll try to  "commit your changes that allow it to compile" then. Thanks a lot in advance!

drtrigon, 2018-12-31:
ping @q3k: what's the status? can I help you with anything?

drtrigon, 2018-12-31:
?? ... I am very puzzeled by your action. As stated in the beginning I am willing to invest my time but need your support. I do have the hardware for tests.

I only closed the issue by accident. We had progress here - I am just waiting for an example by @q3k.

Given the popularity the iCE40 platform has at the moment (Alhambra, TinyFPGA, iCEBreaker, Fomu, etc) I consider supporting it as an important step.

If you prefer to trash these efforts I guess FPGArduino will die at some point in the future anyway. The first sign was that the download links stopped working for weeks recently...

emard, 2018-12-31:
HI

I'm supporting FPGArduino and keep your picorv32 efforts as normal.
We have simple serial bootloader protocol which if you happen to get
running on picorv32 it should be enough to get the board load and run
any other FPGArduino code.

In the meantime I'm also trying to add usb-serial from opencores
as f32c soc which would enable 10Mbit USB 1.1 serial port with a
minimum hardware (2 fpga pins, only cable and few resistors and diodes,
no ftdi chips no usb transcievers)

emard, 2018-12-31:
We have ECP5 board which also supports opensource tools
and picorv32 is supposed to compile on this, I prefer to develop on this
and results will be compatible with ICE40 boards too

So I'd like to call for efforts that someone prepares picorv32 for
ECP5 preferrably with SDRAM support (if won't work then BRAM only)
and get our sime C bootloader here

https://github.com/f32c/f32c/tree/master/src/boot/sio

It should print "rv32>" prompt at terminal on ULX3S or similar ECP5 board
using some usb-serial hardware like ft232r, ft231x or my latest usb core

https://github.com/emard/fpga-usbserial

I will make FPGArduino adaptation to use picorv32 for blinkled at least

drtrigon, 2019-01-01:
@emard: Thanks for your info. As soon as I get some sources I can start testing with, I'll do so. I have to say I don't understand the differences between picorv32 and your f32c-rv32 (SOC). To me it is not important which soft-core/soc is used as long as it works with FPGArduino. (picorv32 is just the thing I am familiar with)
Do I understand correct that ECP5 board code should work one iCE40 boards also? If you can point me into the right direction I'll happy to test this! (do we need the picorv32 if that works?)

One thing to keep in mind is that it would be nice if the soft-core/soc could later be implemented in icestudio also and by that allow to add additional logic easily. From my understanding that should be no big deal. Breaking the soc down into it's parts in icestudio would allow to gain more insight and result in more ways to modify it according to user wishes.

emard, 2019-01-01:
The ECP5 code, if written in portable way, using only generic constructs
and avoiding use of features that ECP5 has and ICE40 doesn't have
then ECP5 code should be reusable for ICE40
with a minimum effort (only clock generator and pin names would differ)

picorv32 should be such case

emard, 2019-01-09:
HI dr3gon

Can you help or get this running on your ICE40 board?
Bootloader works but arduino's code won't execute.

I have compiled picorv32 with opensource tools for my ECP5 board
and replaced its firmware with f32c bootloader (HEX mode only) that
shows rv32> prompt and blinks its leds during upload with

ujprog -P/dev/ttyUSB0 -ra /tmp/arduino_build_668623/BlinkPicoRV32.ino.hex

Blink is arduino sketch:

#define LED (*(volatile uint32_t*)0x02000000)
int ledState = 0;
int previousMillis = 0;
const long interval = 100000;
void setup() {
  LED = 0xFF;
}
void loop() {
  static int currentMillis = 0;
  currentMillis++;
  if (currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;
    if (ledState == 0)
      ledState = 0xFF;
    else
      ledState = 0;
    LED = ledState;
  }
}

emard, 2019-01-09:
The picorv32 and bootloader is here
https://github.com/emard/prjtrellis-picorv32

q3k, 2019-01-09:
I _will_ get you all I know about how to get f32c to synthesize&pr under verific+yosys+nextpnr for ice40 - I just haven't had time to get to it yet :(.

drtrigon, 2019-01-13:
Thanks a lot emard this is great news! Took the code from your repo and modified `makefile` and `scripts/ulx3s_trellis.mk` in order to point to my toolchain:

> F32C-COMPILER-PATH=../toolchain-fpgarduino/f32c/bin

and

> YOSYS ?= $(HOME)/.icestudio/apio/packages/toolchain-icestorm/bin/yosys

finally running `make all` yields this:

```
~/Downloads/icestudio/Alhambra_picorv32-picosoc/prjtrellis-picorv32-master$ make all
$HOME/.icestudio/apio/packages/toolchain-icestorm/bin/yosys \
        -p "hierarchy -top top" \
        -p "synth_ecp5 -noccu2 -json dvi.json" \
        top.v attosoc.v picorv32.v simpleuart.v 

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7 (Apio build) (git sha1 8c071a2, gcc 4.8.4-2ubuntu1~14.04.3 -fPIC -Os)

-- Parsing `top.v' using frontend `verilog' --

1. Executing Verilog-2005 frontend.
Parsing Verilog input from `top.v' to AST representation.
Generating RTLIL representation for module `\top'.
Successfully finished Verilog frontend.

-- Parsing `attosoc.v' using frontend `verilog' --

2. Executing Verilog-2005 frontend.
Parsing Verilog input from `attosoc.v' to AST representation.
Generating RTLIL representation for module `\attosoc'.
Generating RTLIL representation for module `\picosoc_regs'.
Successfully finished Verilog frontend.

-- Parsing `picorv32.v' using frontend `verilog' --

3. Executing Verilog-2005 frontend.
Parsing Verilog input from `picorv32.v' to AST representation.
Generating RTLIL representation for module `\picorv32'.
Generating RTLIL representation for module `\picorv32_regs'.
Generating RTLIL representation for module `\picorv32_pcpi_mul'.
Generating RTLIL representation for module `\picorv32_pcpi_fast_mul'.
Generating RTLIL representation for module `\picorv32_pcpi_div'.
Generating RTLIL representation for module `\picorv32_axi'.
Generating RTLIL representation for module `\picorv32_axi_adapter'.
Generating RTLIL representation for module `\picorv32_wb'.
Successfully finished Verilog frontend.

-- Parsing `simpleuart.v' using frontend `verilog' --

4. Executing Verilog-2005 frontend.
Parsing Verilog input from `simpleuart.v' to AST representation.
Generating RTLIL representation for module `\simpleuart'.
Successfully finished Verilog frontend.

-- Running command `hierarchy -top top' --

5. Executing HIERARCHY pass (managing design hierarchy).

5.1. Analyzing design hierarchy..
Top module:  \top
Used module:     \attosoc
Used module:         \simpleuart
Used module:         \picorv32

5.2. Executing AST frontend in derive mode using pre-parsed AST for module `\picorv32'.
Parameter \BARREL_SHIFTER = 0
Parameter \COMPRESSED_ISA = 0
Parameter \ENABLE_MUL = 1
Parameter \ENABLE_DIV = 0
Parameter \ENABLE_IRQ = 0
Parameter \ENABLE_IRQ_QREGS = 0
Parameter \PROGADDR_RESET = 0
Parameter \PROGADDR_IRQ = 0
Parameter \STACKADDR = 32768
Generating RTLIL representation for module `$paramod$01fe0ba9b03a5d6bd76e78007a69836c7928969b\picorv32'.

5.3. Analyzing design hierarchy..
Top module:  \top
Used module:     \attosoc
Used module:         \simpleuart
Used module:         $paramod$01fe0ba9b03a5d6bd76e78007a69836c7928969b\picorv32
Used module:             \picorv32_pcpi_mul

5.4. Analyzing design hierarchy..
Top module:  \top
Used module:     \attosoc
Used module:         \simpleuart
Used module:         $paramod$01fe0ba9b03a5d6bd76e78007a69836c7928969b\picorv32
Used module:             \picorv32_pcpi_mul
Removing unused module `\picorv32_wb'.
Removing unused module `\picorv32_axi_adapter'.
Removing unused module `\picorv32_axi'.
Removing unused module `\picorv32_pcpi_div'.
Removing unused module `\picorv32_pcpi_fast_mul'.
Removing unused module `\picorv32_regs'.
Removing unused module `\picorv32'.
Removing unused module `\picosoc_regs'.
Removed 8 unused modules.

-- Running command `synth_ecp5 -noccu2 -json dvi.json' --
ERROR: No such command: synth_ecp5 (type 'help' for a command overview)
make: *** [dvi.json] Fehler 1
~/Downloads/icestudio/Alhambra_picorv32-picosoc/prjtrellis-picorv32-master$ 
```

...so what is the proper substitute for `synth_ecp5` ?

emard, 2019-01-13:
HI
synth_ecp5 you should replace by synth_ice40 or similar
ECP5 is my FPGA, ICE40 is yours

drtrigon, 2019-01-14:
@emard: ;) ...ok - thanks! Was not aware of such a obvious replacement. Added.

My current `synth_ice40` (came with icestudio) seems to be old as it does not have the `-json` option. Is it possible to generate the bitstream etc. w/o using the json output? If not, I'll need some time to get a current toolchain... (cuurently it ran until and finished step 6.28 - then make complains about finding no rule for creating `empty_lfe5u-85f.config`).

@q3k: As the ice40 toolchain comming with icestudio contains `arachne-pnr` I'll go and use this until something forces me to change (errors/incompatibilities, size of final design, etc.). Do you remember why you used `nextpnr`? Was there a reason that forced you to use it? Or was it just by accident (because you had it installed already, you where familiar with it, etc.)?

emard, 2019-01-15:
> Have made a fork, and have it close to working with iCE40 on the 'myStorm
> BlackIce II':
>
> https://github.com/uXeBoy/prjtrellis-picorv32
>
> When I hit enter, instead of 'rv32>' I get five garbage characters back, so
> maybe just COM port settings / parity etc need adjusting on the PC side?

Check this serial port speed divider I have something
like 412/2 for 25MHz, correct this factor for your clock
and let it be 115200

uXeBoy, 2019-01-15:
Working now! Was definitely something to do with the built-in serial, but not sure what exactly? Switched to using [one of these](https://www.adafruit.com/product/954) instead and it's all good:

https://raw.githubusercontent.com/uXeBoy/prjtrellis-picorv32/master/blink.gif

emard, 2019-01-16:
> Yes, not all serial cables are reliable and HEX protocol has
> no checksum so it must be 100% reliable.
>
> If that works next step would be to take our f32c binary
> bootloader protocol, it will make it longer but such will
> retry transfer so any bad cable will become useful

HI dr3Gon!

Get new FPGArduino, it will be slightly more friendly
to the picorv32. See README.md in my
https://github.com/emard/prjtrellis-picorv32

drtrigon, 2019-01-17:
Thanks for all the hints so far! Now I had a bit more time to look into it. I started a new Makefile https://github.com/drtrigon/prjtrellis-picorv32/commit/dd55eb30205ece8b66d44a28465ced763f637841 and now I use blif instead of json backend, so the yosys step passes properly. However the arachne-pnr step throws: `fatal error: failed to place: placed 32 RAMs of 68 / 32` so it does not fit into the available BRAM. From manpage http://manpages.org/arachne-pnr I do not see any parameters that allow optimization, so what to remove or disable? Other question is how to connect the clock signal properly, I see `clk_25mhz` but the Alhambra 2 has a 12 MHz clock signal, see https://github.com/drtrigon/prjtrellis-picorv32/commit/f028af7b09b60cf591558b0f1e9086c0b5901255.

I add here the output of the failing step:
```
arachne-pnr -d 8k -P tq144:4k -o top.asc -p demo.pcf top.blif
seed: 1
device: 8k
read_chipdb +/share/arachne-pnr/chipdb-8k.bin...
  supported packages: cb132, cb132:4k, cm121, cm121:4k, cm225, cm225:4k, cm81, cm81:4k, ct256, tq144:4k
read_blif top.blif...
prune...
read_pcf demo.pcf...
demo.pcf:21: warning: no port `clk' in top-level module `top', constraint ignored.
demo.pcf:24: warning: no port `flash_csb' in top-level module `top', constraint ignored.
demo.pcf:25: warning: no port `flash_miso' in top-level module `top', constraint ignored.
demo.pcf:26: warning: no port `flash_mosi' in top-level module `top', constraint ignored.
demo.pcf:27: warning: no port `flash_clk' in top-level module `top', constraint ignored.
instantiate_io...
pack...

After packing:
IOs          19 / 107
GBs          0 / 8
  GB_IOs     0 / 8
LCs          2852 / 7680
  DFF        647
  CARRY      433
  CARRY, DFF 126
  DFF PASS   224
  CARRY PASS 17
BRAMs        68 / 32
WARMBOOTs    0 / 1
PLLs         0 / 2

place_constraints...
promote_globals...
  promoted clk_25mhz$2, 909 / 909
  promoted $abc$27963$n5, 435 / 435
  promoted $abc$27963$n2173, 66 / 66
  promoted $abc$27963$n2227, 64 / 64
  promoted $abc$27963$n2163, 52 / 53
  promoted $abc$27963$n2161, 49 / 49
  promoted $abc$27963$n3, 31 / 31
  promoted $abc$27963$n1, 6 / 6
  promoted 8 nets
    3 sr/we
    4 cen/wclke
    1 clk
  8 globals
    3 sr/we
    4 cen/wclke
    1 clk
realize_constants...
  realized 1
place...
fatal error: failed to place: placed 32 RAMs of 68 / 32
```

drtrigon, 2019-01-17:
Btw.: I would also like to test it with https://github.com/f32c/f32c but when I looked into it I was a bit overwhelmed by the options and variants that are in there. Do I understand correct, it only exists in VHDL but not in Verilog? Thus I would first have to setup the yosys plugin (and by that next-pnr too according to http://www.clifford.at/icestorm/) or better the whole icestorm toolchain as icestudio contains a yosys w/o plugin support (and w/o json output).

drtrigon, 2019-01-17:
@emard: OK, managed to make it synthesize by reducing memory usage in `attosoc.v`with `MEM_WORDS = 2048` https://github.com/drtrigon/prjtrellis-picorv32/commit/557f71a5da9eb6f4d5882aa9cf2028efbc7455db. Now it takes 20 of 32 BRAM blocks. However I have have no idea whether that's ok? As soon as you can give me feedback about MEM_WORDS, the clock signal and the baudrate setting I'll upload and test it!

uXeBoy, 2019-01-17:
@drtrigon - to use the 12MHz clock you will need to change `reg_uart_clkdiv = 416/2` in **firmware.c** to `reg_uart_clkdiv = 104` (12,000,000 / 115200 = 104.166666667)

@emard - the reason for my issue with serial has been found, `reg_uart_clkdiv` should be 217 to use with 25MHz, not 208 (416/2)

drtrigon, 2019-01-17:
Ok, changed the setting as suggested and just connected clk with clk_25mhz, see https://github.com/drtrigon/prjtrellis-picorv32/commit/41b7980cb1180c3c95450619d8c589d3effacb5c. Uploaded it, but no blinking leds (yet)... have tried to set https://github.com/drtrigon/prjtrellis-picorv32/blob/master/firmware.c#L62 to 1 in order to enable it, but no change... suggestions? ideas?

emard, 2019-01-17:
My apologies, this baud rate math was slightly off,
but yes you got it right
For start you can feed 12 MHz to picorv32 just to make
sure it works, later rise the clock until it stops working and
back off 10-15% to be safe for normal use

when memory is shorter: 2 things need to be changed:
1. in firmware.c set lower the stack address 7FF0 means
16 bytes below 32K. change it similar for your smaller mem
                       "li s0, 0x00007FF0;"    /* stack address */

2. in sections.lds also you should lower the length similar
    RAM (xrw)       : ORIGIN = 0x00000000, LENGTH = 0x008000 /* 32 KB */

drtrigon, 2019-01-17:
@emard: Thanks for the hints. If I got the math right, I have for 1/4 of the memory size (8K) to set `hex(2048*4-16) = 1FF0` instead of `7FF0` and `hex(2048*4) = 0x002000` instead of `0x008000` as done in https://github.com/drtrigon/prjtrellis-picorv32/commit/623c3a6cfb72f7679171b736a32358003e6ef2ce but still no reaction...

I don't understand your comment about rising the clock. At the moment I just connected the 12MHz to your `clk_25mhz` signal. Do I have to adopt dividers for that also?

emard, 2019-01-18:
HI

Great, seems you calculated 8K RAM constants well
I have rewritten readme with intention to solve FAQ around here.
It's working on my board and latest FPGArduino

I can change RAM to see if it works or my README is maybe
not telling the complete truth

No reaction, no rv32> boot prompt on serial console?
check that you have correctly connected TX/RX

In my example I'm connecting RX of FPGA to TX of USB-SERIAL
and vice versa TX of FPGA to RX of USB-SERIAL

emard, 2019-01-18:
In my latest git there's firmware.c with proper integer
math for the divisor. Just change the clock 12 MHz = 12000000 Hz
compile, it will calculate divisor automatically

drtrigon, 2019-01-18:
@emard: Thanks again for your hints and efforts. Currently I am stuck as I must be doing something fundamentally wrong. I double checked the RX, TX connections, swapped them several times, there is no prompt appearing. What happens is when I swap RX/TX I get some leds on like 0b10101010. When I enable the stuff on line 62 https://github.com/drtrigon/prjtrellis-picorv32/blob/master/firmware.c#L62 I get an led pattern like 0b00001111. Adding LED = 0xFF; in firmware.c right before prompt label does not change anything. I never get serial output. Summary something happens but I have no clue what currently...

emard, 2019-01-18:
Well your thing seems to work, don't enable this stuff #if 0
it was just for debugging
but led pattern you discribed is OK so at least you compiled
successfully your code

I think you are close, Either serial port or pinout is wrong or
baudrate or clock rate, something simple

0b10101010 is normal startup pattern that's good. and
0b00001111 from #if 1 enabled is indication that bootloader
is executing C compiled code, great.

when this #if 0 stuff is not enabled, any key press on the prompt
should show on the LED the binary form of the ascii letter you pressed

Im no expert on ICE40 field, but I know picorv32 was initially developed
for ICE40 so it should work  there.

emard, 2019-01-18:
> @emard: Thanks a lot for the feedback - I did not find where in code these
> LED patterns are set - if you say that's a indication of running code,
> that's a very good sign. Thanks!
>
> I have several icestudio projects using the same picorv32 and they work -
> that is where the idea for this feature request comes from. I can provide
> them also later e.g. as I have implemented GPIO (read and write) including
> tri-state port etc. - that's a nice topic for later.
>
> Coming back to the current problem; by using `LED = 0xFF;` as "breakpoint"
> replacement I am trying to track down where my code hangs. What I found is
> that is does not pass beyond
> https://github.com/drtrigon/prjtrellis-picorv32/blob/master/firmware.c#L78
> (does not leave the do while after the loop label). To me it looks like
> serial issue. The divider (104) is ok, I used the same in the other
> mentioned projects successfully. The thing is I double checked the TX/RX
> signals again and am 99% sure that  they are wrong now - but if I swap them,
> no C code gets executed at all (`LED = 0xFF;` in very first line does
> nothing)... ideas?

HI

This is strange, maybe a bad compilation or too congested design for
small ICE40.
I can ask somebody who has ICE40 to try it too

To verify  tx and rx I woulld do in toplevel something like
assign tx = rx;
so if I connect with serial console, whatever I type should be echoed back,
that is the test that tx and rx and serial port are working.

There is also "helloworld.c" example that you can copy over
firmware.c and see what happens.

I have further updated my project by adding serial break
detection to the simpleuart.v and I it works (FPGArduino sends
serial break to reset CPU back to bootloader)

drtrigon, 2019-01-18:
@emard: Ok following improvement; I pulled your latest code and merged. Now I see a `rv32>` prompt after reset! What puzzles me is that the leds do not react to or show the value of the last byte sent.

drtrigon, 2019-01-18:
@emard: Testting the `helloworld.c` now ... sending from FPGA to PC works but it seems the back path does not ... have to investigate that...

emard, 2019-01-18:
rv32> prompt is good, if LEDs don't follow typing
I would suspect bad FPGA synthesys, upgrade your ICE40 tools.

Also when you have rv32> prompt, you can immediately
try sending a binary from FPGArduino. Follow my README.md
and upload the blink

emard, 2019-01-18:
rv32> should appear each time you press ENTER
not only at reset, also if you send BREAK the prompt will reappear
suppose you use terminal
screen /dev/ttyUSB0 115200
break is sent by ctrl-a ctrl-b

drtrigon, 2019-03-17:
Had some time to give it a try. Installed the fpgarduino toolchain successfully using the boards-manager method. Now the newest arachne-pnr refuses to route:

```
$ make
arachne-pnr -d 8k -P tq144:4k -o top.asc -p demo.pcf top.blif
seed: 1
device: 8k
read_chipdb +/share/arachne-pnr/chipdb-8k.bin...
  supported packages: bg121, bg121:4k, cb132, cb132:4k, cm121, cm121:4k, cm225, cm225:4k, cm81, cm81:4k, ct256, tq144:4k
read_blif top.blif...
prune...
read_pcf demo.pcf...
note: no set_io constraints for pins `btn[0..6]'
note: no set_io constraints for pin `wifi_gpio0'
demo.pcf:32: fatal error: Missing 8 set_io constraints
make: *** [top.asc] Error 1
```
where the older one just did this:
```
promote_globals...
  promoted clk_25mhz$2, 811 / 811
  promoted $abc$27162$n7, 287 / 287
  promoted $abc$27162$n5, 147 / 147
  promoted $abc$27162$n2040, 66 / 66
  promoted $abc$27162$n2095, 64 / 64
  promoted $abc$27162$n2030, 52 / 53
  promoted $abc$27162$n2028, 59 / 59
  promoted $abc$27162$n3, 31 / 31
  promoted 8 nets
```
have to figure that out now ...

[43 more replies]