f32c/arduino #33: undefined reference to operator delete

uXeBoy, 2019-11-25:
Getting this error during compilation - seems to be related to missing libstdc++

...is there a possible solution??

emard, 2019-11-26:
I'm getting on some projects similar "delete" missing c++ issue
I think I workarounded this somehow but I would like if
"gornjas" or somebody would propose a better solution to this...

gornjas, 2019-11-26:
If you have problems with operator delete, you should also have problems with operator new. Both are implemented in hardware/fpga/f32c/cores/f32c/new.cpp.

A simple, minimal example which can reliably provoke the error could help us pinpoint the exact problem.

uXeBoy, 2019-11-26:
I think I see the problem now... 🤦

That **new.cpp** does not include C++14 style _'sized deallocation'_ functions:

https://en.cppreference.com/w/cpp/memory/new/operator_delete

gornjas, 2019-11-26:
Hmm as a person who uses or cares about C++ only when shotgun-forced, I can only comment that our new.cpp was blindly cloned many years ago from (I believe):
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/new.cpp

Does your application compile with AVR arduino?

uXeBoy, 2019-11-26:
Did some research, and I ended up just adding `-fno-sized-deallocation` to the c++ compiler flags in platform.txt and that fixed it... maybe a better solution if the the added delete variants would just be dummies anyway?

gornjas, 2019-11-26:
I like your approach with patching compiler flags in platform.txt better, as there's no point in introducing more dummy functions which are equivalent to the existing ones, and just occupy space.

I'd leave it to Davor to check whether adding the proposed flag in platform.txt does any harm, as he's using fpgarduino much more frequently than I am, and if it doesn't break anything to proceed and commit the change.